Project-Aloha/mu_aloha_platforms #31: Confused by step 4 in the build instructions in MU-sm8150 porting guide

49b, 2022-09-22:
[Tutorial](https://renegade-project.org/#/en/sm8150pkg/tutorial/3_Build_instructions?id=step-4-configure-the-build-system-to-process-the-firmware-binaries-for-your-device)

This is what it says in step 4:
> We will now need to inform the build system of these files. The files DXE0.inc through DXE3.inc tell the build system about the locations of these firmware binaries for each separate device. So, we will make a copy of one of these files and modify it with the correct paths for our own files.
>
> Let's copy it as follows:
>
> ```
> (SurfaceDuo) $ cp Platforms/SurfaceDuoPkg/Include/DXE3.inc Platforms/SurfaceDuoPkg/Include/DXE4.inc
> ```
> Then, we'll open it in Nano:
>
> ```
> (SurfaceDuo) $ nano Platforms/SurfaceDuoPkg/Include/DXE4.inc
> ```
> In nano, we will find the following string:
>
> ```
> SECTION PE32 = SurfaceDuoPkg/CustomizedBinaries/XiaoMi/Andromeda_MIX3_5G/
> ```

But there is no `Platforms/SurfaceDuoPkg/Include/DXE3.inc`, so can anyone please help? thanks.

gus33000, 2022-09-22:
The guide you are currently following is outdated due to numerous changes that have been made ever since. On top of that, there is also an ongoing rebase that you can watch happening under other specific sub branches. If it's not urgent, I'd suggest waiting til the repository settles down back into a stable state. Out of curiosity, which device are you trying to add support with? Plenty of devices are already supported at the moment.

49b, 2022-09-22:
> The guide you are currently following is outdated due to numerous changes that have been made ever since. On top of that, there is also an ongoing rebase that you can watch happening under other [...]

Samsung Galaxy Note 10 Plus (5G)

gus33000, 2022-09-22:
That device seems already supported under ````https://github.com/edk2-porting/MU-sm8150pkg/tree/main/Platforms/SurfaceDuoPkg/Device/samsung-beyond1qlte````. Might need a slight adaptation for the plus model. However UFS is not functional under Windows at all.

gus33000, 2022-09-22:
No, which is why I mentioned you'll need to adapt this for the Plus model above.

49b, 2022-09-23:
Hi, I am adapting to Note 10 Plus 5G (started by copying S10's folder `Platforms/SurfaceDuoPkg/Device/samsung-beyond1qlte`) and I am changing the contents of the copied folder to work with Note 10 Plus 5G, but before modifying I would like to know, is it safe to use S10's `PatchedBinaries/UFSDxe.efi` (Not same as `CustomizedBinaries/UFSDxe.efi` according to sha1 hashes) and `ACPI/DSDT.aml` files for Note 10 Plus 5G? thanks.

gus33000, 2022-09-23:
No.

Idonotkno, 2022-09-23:
@49b Instead of using ufsdxe in the shared configuration, Samsung's ufsdxe has been patched, and you can use S10 patched ufsdexe instead of something else, or you might brick your phone.
But do not use the original ufsdxe, Samsung devices have been specified after the patch of the ufs driver, please do not try other.

sunflower2333, 2022-09-24:
you can use docker :)
And the porting Guide has been deserted for a long time.

49b, 2022-11-24:
> @49b Instead of using ufsdxe in the shared configuration, Samsung's ufsdxe has been patched, and you can use S10 patched ufsdexe instead of something else, or you might brick your phone. But do not [...]

Hi, Using Samsung Galaxy S10's patched UFSDxe did not work on my Samsung Note 10 Plus, so do you know how to build a patched UFSDxe driver, please? The tutorial on patching drivers in edk2-porting/renegade-project did not say anything about patching UFSDxe. Thanks.

sunflower2333, 2022-11-24:
Hi,
 here<https://github.com/woa-msmnile/MU-sm8150pkg/blob/main/Documentation/SimpleGuide.md#part-3-patch-binaries> is a simple notice. You can try to check the difference between the patch one and the original one of Samsung S10. And apply the same patch to your ufsdxe.

Sunflower2333.

49b, 2022-11-24:
Hi, Thanks for responding. There is a step I am confused in, when I do `diff a.txt b.txt`, there is many lines, unlike a few lines like the one in the tutorial, and I cannot tell the different adresses to modify. If I run the following commands:
```
hexdump -C MU-sm8150pkg/Platforms/SurfaceDuo1Pkg/Device/samsung/CustomizedBinaries/UFSDxe.efi > a.txt # (Original binary) 
hexdump -C MU-sm8150pkg/Platforms/SurfaceDuo1Pkg/PatchedBinaries/UFSDxe.efi > b.txt # (Surface duo patched binary)
```
then doing either `diff a.txt b.txt` will return many lines, unlike the tutorial. is there any mistake I am doing? Thanks.

sunflower2333, 2022-11-24:
A lot of difference? Sounds like it was patched a lot. Hard work. Good luck. Btw , can you send me the difference ?( diff xxx xxx > mydiff.txt)

Sunflower2333

49b, 2022-11-24:
Hi, Thanks for responding. here is the diff file: [diff.txt](https://github.com/woa-msmnile/MU-sm8150pkg/files/10085213/diff.txt)

sunflower2333, 2022-11-24:
***@***.***Looks like you are comparing the patched s10’s ufsdxe with Duo’s.
You should compare with S10’s original UFSDXE

Sunflower2333

49b, 2022-11-24:
Hi, here is the file showing the diff between Original Samsung S10 and Patched Samsung S10 UFSDxe.efi files: [diff.txt](https://github.com/woa-msmnile/MU-sm8150pkg/files/10085350/diff.txt)
Thanks.

sunflower2333, 2022-11-24:
***@***.***Not the same version ? one is 8814, the other is 8815. Weird.

Sunflower2333

ammmarameen1, 2023-10-08:
Hi @49b have you succeeded in running win 11 on samsung galaxy note 10+ 5G?  I have basic programming skill, and I want to know the possibility of having dual boot on my note 10+ 5G snapdragon.