Project-Aloha/mu_aloha_platforms #32: edk2 unable to boot on vayu

halal-beef, 2022-10-01:
after building, i tried to boot the image and im met with loads of texts and then a prompt reboot into android, i have tried to change dtb and the CustomizedBinaries folder with up to date ones, but still no avail

heres the only picture of what was happening before crash / reboot (sadly not the clearest image)

[image]

halal-beef, 2022-10-01:
last driver to load was haliommu

degdag, 2022-10-01:
> last driver to load was haliommu

It's not the last driver loaded but it's PilDxe .
You can comment it from DXE.inc

halal-beef, 2022-10-01:
oops my bad, alright ill comment it out

degdag, 2022-10-01:
> oops my bad, alright ill comment it out

However if you plan to use winpe , than USB will not work on pe even with a powered hub
I didn't tested surface duo guide but I guess you could try the guide if you plan to test the full windows

halal-beef, 2022-10-01:
we are using another method for winpe, making a pe partition then installing pe to there

degdag, 2022-10-01:
> we are using another method for winpe, making a pe partition then installing pe to there

Even if you boot to winpe you can't use mouse or keyboard over USB to deploy the full windows installation .
However as I told you . you can try surface duo installation guide if you want full windows installation

halal-beef, 2022-10-01:
thanks for the warning atleast

halal-beef, 2022-10-01:
tysm the efi now boots

closed 2022-10-01 by halal-beef