Project-Aloha/mu_aloha_platforms #47: Error building with `./build_uefi.sh`

49b, 2022-11-03:
Hi, when building with `./build_uefi.sh -d [target device] -s [ram size]`, I get an error. I am not sure what error is that, but here is the log. Does anyone know how to fix the build error? Thanks.
```
INFO - /home/linuxuser/Documents/MU-sm8150pkg/MU_BASECORE/MdeModulePkg/Library/ResetUtilityLib/ResetUtility.c:19:13: error: field ResetSubtype within 'RESET_UTILITY_GUID_SPECIFIC_RESET_DATA' is less aligned than 'GUID' and is usually due to 'RESET_UTILITY_GUID_SPECIFIC_RESET_DATA' being packed, which can lead to unaligned accesses [-Werror,-Wunaligned-access]
INFO -   GUID      ResetSubtype;
INFO -             ^
INFO - 1 error generated.
INFO - make: *** [GNUmakefile:295: /home/linuxuser/Documents/MU-sm8150pkg/Build/samsung-d2x-AARCH64/DEBUG_CLANG38/AARCH64/MdeModulePkg/Library/ResetUtilityLib/ResetUtilityLib/OUTPUT/ResetUtility.obj] Error 1
INFO - 
INFO - 
INFO - build.py...
INFO -  : error 7000: Failed to execute command
INFO - 	m a k e   t b u i l d [/home/linuxuser/Documents/MU-sm8150pkg/Build/samsung-d2x-AARCH64/DEBUG_CLANG38/AARCH64/MdeModulePkg/Library/ResetUtilityLib/ResetUtilityLib]
INFO - 
INFO - 
INFO - build.py...
INFO -  : error F002: Failed to build module
INFO - 	/home/linuxuser/Documents/MU-sm8150pkg/MU_BASECORE/MdeModulePkg/Library/ResetUtilityLib/ResetUtilityLib.inf [AARCH64, CLANG38, DEBUG]
INFO - 
INFO - - Failed -
INFO - Build end time: 01:30:12, Nov.03 2022
INFO - Build total time: 00:00:11
INFO - 
INFO - ------------------------------------------------
INFO - --------------Cmd Output Finished---------------
INFO - --------- Running Time (mm:ss): 00:11 ----------
INFO - ----------- Return Code: 0x00000001 ------------
INFO - ------------------------------------------------
ERROR - Compiler #7000 from :   Failed to execute command
ERROR - EDK2 #002 from :   Failed to build module
CRITICAL - Build failed
PROGRESS - End time: 2022-11-03 01:30:12.599297	 Total time Elapsed: 0:00:11
SECTION - Log file is located at: /home/linuxuser/Documents/MU-sm8150pkg/Build/BUILDLOG_SurfaceDuo1Pkg.txt
SECTION - Summary
ERROR - Error
```

sunflower2333, 2022-11-03:
I met it too when *debugging* workflow script.  
Seems it didn't appear any more after changing the ubuntu version to 20.04.

sunflower2333, 2022-11-04:
Can you try the `dtb_test` branch, It updated submodules. 
Perhaps this issue caused by outdated submodules.

49b, 2022-11-04:
Hi, because I used step 5 from the Porting guide to build, and not the building guide from the README.md to build, I was missing a step, and that is to use docker to build if the distro is not Ubuntu 20.04. My apologies.

closed 2022-11-04 by 49b