Project-Aloha/mu_aloha_platforms #50: I want to port UEFI to my Pixel 4 XL,but I can't find the porting guide.

Android-Piepaint, 2022-11-07:
I want to port UEFI to my Pixel 4 XL. I cloned the repository,and I don't know how to start. I can't find a build guide, the old one seems to be outdated. If you know where the guide is, please tell me,thanks.

sunflower2333, 2022-11-08:
Hello.
  I do not suggest porting it to pixel. You UFS will be erased if you add UFSDXE. So, it's very dangerous.

  The porting guide is on the way. I can provide a quick note.
  Quick Note:
   1. copy nubia-tp1803 to google-coral. (Brand-codename).
    2. Remove all files under google-coral/CustomizedBinaries & google-coral/PatchedBinaries.
    3. Extract files from your device's xbl and put them under CustomizedBinaries.
    4. Enable MLVM in Defines.dsc.inc
    5. Edit resolution in PcdFixedAtBuild.dsc.inc
    6. Remove UFSDxe part in DXE_Spec.inc.
    7. Patch your device's dxe and put them under PatchedBinaries.
    8. Replace tp1083.dtb with coral.dtb (you can find your device's dtb in /sys/firmware/fdt.)
    9. ./build_uefi.sh -d google-coral -s <ram-size>
    10. fastboot boot Build/google-coral/google-coral_xG.img

Again, if you added UFSDxe, there is a high possibility that  your ufs will be erased.

Sunflower2333.

sunflower2333, 2022-11-08:
1.
  - excute dd if=/dev/block/by-name  of=/sdcard/xbl.img on your phone.
  - copy or move it to your computer.
  - right click -> 7zip- > extract -> #:e -> 4.gz
     - you will see all efi files in 4.gz

2. How to patch:
  - You can use IDA to patch it. I think the old guide has it.
  - To know where needs patch, you need to refer other devices:
    ```
    hexdump -C  a.efi > a.txt
    hexdump -C b.efi > b.txt
    diff a.txt b.txt
    ```
  - a.efi is original binary, b.efi is a patched binary.
  - The diff will tell you where and how to patch.
  - Find the same function in your device's dxe.
  - Patch it with a tool.

3.
  - Nothing to do with your dtb except rename.

4.
  - As you can see in the README.md, if you do not have ubuntu20.04 , you can choose docker. Or, you can use github workflow. I haven't tested other higher verisons or other distros.

5.
  - boot from USB is not available, because the usb mode in uefi is peripheral.
  - boot from SD, I do not know. Maybe SdccDxe needs patch. You can have a try.

Sunflower2333.

Android-Piepaint, 2022-11-08:
> 1. - excute dd if=/dev/block/by-name of=/sdcard/xbl.img on your phone. - copy or move it to your computer. - right click -> 7zip- > extract -> #:e -> 4.gz - you will see all efi files in 4.gz 2. How [...]

That makes me more clearly to build. But I still have 3 questions left:
1.My phone is running LineageOS , can I use Lineage OS's DTB and xbl.img? 
2.If I can't , my phone's stock rom has multiple Android versions, from 10 to 13. Which version I should use to extract DTB and xbl.img?
3.According to step 2, What is IDA? How can I find the dxe in the same function like my device's dxe ? Which tool I need to patch?
I'm sorry to trouble you, but I'd appreciate it if you could help me.

sunflower2333, 2022-11-08:
1. You can use the dtb and the xbl.
3. IDA is the tool. Or you can choose Ghidra.

Android-Piepaint, 2022-11-08:
Thanks for helping me!

sunflower2333, 2022-11-13:
You are welcome. If you do not have other issue, I'll close this issue.

Android-Piepaint, 2022-11-13:
Well, I have finished building,and I got the UEFI image, however, it refuse to boot on my Pixel 4 XL. Why? Did I did something wrong?

Android-Piepaint, 2022-11-13:
[image]
By the way, someone let me do these.

sunflower2333, 2022-11-13:
拒绝启动？你可以fastboot devices看一下启动没启动。如果启动了，fastboot devices是看不到设备的。如果可以看到。。那就是dtb问题了。
至于ufs啊，你需要patch ufsdxe，虽然我不建议。如果patch了，你可以在uefi使用ufs，但是如果你使用uefi启动什么系统（比如win或者linux），然后那些系统上的驱动尝试调用ufs，你的ufs会被擦。对于windows，你需要在dsdt里面删掉，
```
iasl DSDT.aml
```
会生成DSDT.dsl
删除其中的Device(UFS0){XXX}那一大块就好了
然后
```
iasl DSDT.dsl
```
编译回去

Android-Piepaint, 2022-11-13:
Fastboot 提示error boot prepare ，直接boot 的話ADB 會提示無法校驗映像...那真正的DTB到底怎麼提取？

sunflower2333, 2022-11-13:
> Fastboot 提示error boot prepare ，直接boot 的ADB 崾o法校映像...那真正的DTB到底怎N提取？

那还真是稀奇，没见过，/sys/firmware/fdt那个fdt应该是可以启动的

Sunflower2333.

Android-Piepaint, 2022-11-13:
是不是DTB複製到電腦的時候出的問題？好像要用ADB命令複製？

sunflower2333, 2022-11-13:
应该没区别吧，你看看sha1sum呢

Sunflower2333.

是不是DTB}u到X的r候出的}？好像要用ADB命令}u？

Android-Piepaint, 2022-11-13:
那要怎麼看呢？

sunflower2333, 2022-11-13:
你提取一份到电脑上，看看他的sha校验码
然后对手机里面那个用
```
sha1sum fdt
```
这个命令可以得到fdt的校验码。我觉得应该不会有啥问题

Sunflower2333.

那要怎N看呢？

Android-Piepaint, 2022-11-13:
校驗了，是一致的，但就是無法啟動。會不會是重命名方式的問題？

sunflower2333, 2022-11-13:
这个应该不是。。你问问w吧，就酷安上面那位

Sunflower2333.

校了，是一致的，但就是o法印侵孛绞降}？

Android-Piepaint, 2022-11-13:
那一位？

sunflower2333, 2022-11-13:
假装微笑的很好

Sunflower2333.

Android-Piepaint, 2022-11-13:
好的。

closed 2022-11-13 by Android-Piepaint

Thang150898, 2022-11-26:
How to build an image please guide in detail

sunflower2333, 2022-11-26:
Hi,
I think the readme.md tells all the steps your need to do for compiling. If you meet any issues while executing cmds in README.md, report it here, thanks.

Sunflower2333.

sunflower2333, 2022-11-26:
also, you can download the lasted built images in  Github Action.

Sunflower2333.