Project-Aloha/mu_aloha_platforms #110: docker build -t mu:v1 .

Ost268, 2023-09-03:
Reading package lists...
W: https://packages.microsoft.com/ubuntu/20.04/prod/dists/focal/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
E: The repository 'https://download.mono-project.com/repo/ubuntu stable-jammy Release' does not have a Release file.

sunflower2333, 2023-09-03:
@Ost268 git pull and try the lastest commit.

Ost268, 2023-09-03:
root@20eg:/home/vboxuser/msmnilePkg# pip install --upgrade -r pip-requirements.txt
ERROR: Could not find a version that satisfies the requirement edk2-pytool-library~=0.14.1 (from -r pip-requirements.txt (line 15)) (from versions: 0.9.0, 0.9.1, 0.9.2, 0.10.0, 0.10.1, 0.10.2, 0.10.3, 0.10.4, 0.10.5, 0.10.6, 0.10.7, 0.10.8, 0.10.9, 0.10.10, 0.10.11, 0.10.12, 0.10.13, 0.10.15, 0.11.0, 0.11.1, 0.11.2, 0.11.4, 0.11.5, 0.11.6, 0.12.0, 0.12.1, 0.15.0, 0.15.1, 0.15.2, 0.15.3, 0.15.4, 0.16.0, 0.16.1, 0.16.2, 0.17.0)
error edk2-pytool-library 14.1 work on 15.1 and later

ERROR: No matching distribution found for edk2-pytool-library~=0.14.1 (from -r pip-requirements.txt (line 15))

Ost268, 2023-09-03:
``not a big mistake, but for new users who don"t know how to fix is a big problem, sorry for persistence Mr.sunflower2333
The project is developing and new users who want Windows on the phone are increasing, thank you and others for support and development of thhe project

closed 2023-09-03 by Ost268

sunflower2333, 2023-09-03:
@Ost268 thanks, but i'm very sure that i run this command successfully 4 hours ago. Perhaps 0.14.x was removed just now. You can bump it to 0.15.3 and have a try.

Ost268, 2023-09-03:
SECTION - Initial update of environment
UpdatingWARNING - [SDE] Failed to fetch NugetDependecy: edk2-basecrypto-driver-bin@2022.08.6: [Nuget] We failed to install this version 2022.08.6 of edk2-basecrypto-driver-bin
WARNING - [SDE] Failed to fetch NugetDependecy: mu-uncrustify-release@73.0.8: [Nuget] We failed to install this version 73.0.8 of mu-uncrustify-release
WARNING - [SDE] Failed to fetch NugetDependecy: mu_nasm@20016.1.1: [Nuget] We failed to install this version 20016.1.1 of mu_nasm
WARNING - [SDE] Failed to fetch NugetDependecy: edk2-acpica-iasl@20210105.0.6: [Nuget] We failed to install this version 20210105.0.6 of edk2-acpica-iasl
....WARNING - [SDE] Failed to fetch NugetDependecy: Mu-Basetools@2022.08.4: [Nuget] We failed to install this version 2022.08.4 of Mu-Basetools
. Done
SECTION - Second pass update of environment
UpdatingWARNING - [SDE] Failed to fetch NugetDependecy: Mu-Basetools@2022.08.4: [Nuget] We failed to install this version 2022.08.4 of Mu-Basetools
WARNING - [SDE] Failed to fetch NugetDependecy: edk2-acpica-iasl@20210105.0.6: [Nuget] We failed to install this version 20210105.0.6 of edk2-acpica-iasl
WARNING - [SDE] Failed to fetch NugetDependecy: mu_nasm@20016.1.1: [Nuget] We failed to install this version 20016.1.1 of mu_nasm
WARNING - [SDE] Failed to fetch NugetDependecy: edk2-basecrypto-driver-bin@2022.08.6: [Nuget] We failed to install this version 2022.08.6 of edk2-basecrypto-driver-bin
WARNING - [SDE] Failed to fetch NugetDependecy: mu-uncrustify-release@73.0.8: [Nuget] We failed to install this version 73.0.8 of mu-uncrustify-release
. Done
ERROR - We were unable to successfully update 5 dependencies in environment
SECTION - Summary
ERROR - Error

ERROR - Dependency 'edk2-basecrypto-driver-bin' is not met!
DEBUG - Verify 'Mu-Basetools' returning 'False'.
DEBUG - Verify 'Mu-Basetools' returning 'False'.
ERROR - Dependency 'Mu-Basetools' is not met!
DEBUG - Verify 'edk2-acpica-iasl' returning 'False'.
DEBUG - Verify 'edk2-acpica-iasl' returning 'False'.
ERROR - Dependency 'edk2-acpica-iasl' is not met!
DEBUG - Verify 'mu_nasm' returning 'False'.
DEBUG - Verify 'mu_nasm' returning 'False'.
ERROR - Dependency 'mu_nasm' is not met!
DEBUG - Verify 'mu-uncrustify-release' returning 'False'.
DEBUG - Verify 'mu-uncrustify-release' returning 'False'.
ERROR - Dependency 'mu-uncrustify-release' is not met!

sunflower2333, 2023-09-03:
@Ost268 i guess nuget is not installed.

sunflower2333, 2023-09-03:
@sunflower2333 apt install nuget then retry?

Ost268, 2023-09-03:
Thank you for your support.  Galaxy Fold in win pe min acpi works usb, 1 core, and xhci usb hub does not work in Windows

sunflower2333, 2023-09-03:
@Ost268 why 1 core, i remember nobody disabled cores for sdm855 in acpi.

sunflower2333, 2023-09-03:
@Ost268 i still thought it's dangerous to try on samsung devices unless you have ways to unbrick it.

Ost268, 2023-09-03:
sure, I know about it

Ost268, 2023-09-04:
the same error
ERROR - Dependency 'edk2-basecrypto-driver-bin' is not met!
DEBUG - Verify 'Mu-Basetools' returning 'False'.
DEBUG - Verify 'Mu-Basetools' returning 'False'.
ERROR - Dependency 'Mu-Basetools' is not met!
DEBUG - Verify 'edk2-acpica-iasl' returning 'False'.
DEBUG - Verify 'edk2-acpica-iasl' returning 'False'.
ERROR - Dependency 'edk2-acpica-iasl' is not met!
DEBUG - Verify 'mu_nasm' returning 'False'.
DEBUG - Verify 'mu_nasm' returning 'False'.
ERROR - Dependency 'mu_nasm' is not met!
DEBUG - Verify 'mu-uncrustify-release' returning 'False'.
DEBUG - Verify 'mu-uncrustify-release' returning 'False'.
ERROR - Dependency 'mu-uncrustify-release' is not met!

sunflower2333, 2023-09-05:
pip install -r pip-requirements.txt

Sunflower2333.

sunflower2333, 2023-09-05:
I suggest using the test branch(sunflower2333/test/2307) because the submodules/piprequirements may be outdated in main branch.

Sunflower2333.

Ost268, 2023-09-17:
INFO - Log Started: Sunday, September 17, 2023 07:20PM
SECTION - Init Self Describing Environment
DEBUG - --- self_describing_environment.__init__()
DEBUG - Skipped directories specified = ()
DEBUG - --- self_describing_environment.load_workspace()
DEBUG - Loading workspace: /build
DEBUG -   Including scopes: SurfaceDuo1, gcc_aarch64_linux, edk2-build, cibuild, configdata, global-nix, global
DEBUG - --- self_describing_environment._gather_env_files()
DEBUG - Adding descriptor /build/MU_BASECORE/edk2_core_path_env.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/basetools_calling_path_env.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/basetools_path_env.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/BinWrappers/PosixLike/posix_path_env.yaml to the environment with scope global-nix
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Scripts/basetools_scripts_bin_path_env.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Source/Python/basetool_tiano_python_path_env.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/UncrustifyCheck/uncrustify_ext_dep.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Bin/nasm_ext_dep.yaml to the environment with scope edk2-build
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Bin/iasl_ext_dep.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Bin/basetoolsbin_ext_dep.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/CryptoPkg/Driver/Bin/BaseCryptoDriver_ext_dep.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/CompilerPlugin/Compiler_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/SpellCheck/SpellCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/LibraryClassCheck/LibraryClassCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/DscCompleteCheck/DscCompleteCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/CharEncodingCheck/CharEncodingCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/DebugMacroCheck/BuildPlugin/DebugMacroCheck_plug_in.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/DependencyCheck/DependencyCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/UncrustifyCheck/uncrustify_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/LineEndingCheck/line_ending_check_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/MarkdownLintCheck/MarkdownLintCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/ImageValidation/ImageValidation_plug_in.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/GuidCheck/GuidCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/ClangPdbToolChain/ClangPdbToolChain_plug_in.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/FlattenPdbs/FlattenPdbs_plug_in.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/WindowsCapsuleSupportHelper/WindowsCapsuleSupportHelper_plug_in.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/Edk2ToolHelper/Edk2ToolHelper_plug_in.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/OverrideValidation/OverrideValidation_plug_in.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/FdSizeReport/FdSizeReportGenerator_plug_in.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/BuildToolsReport/BuildToolsReportGenerator_plug_in.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/LinuxGcc5ToolChain/LinuxGcc5ToolChain_plug_in.yaml to the environment with scope global-nix
DEBUG - Adding descriptor /build/Features/CONFIG/SetupDataPkg/Plugins/UpdateConfigHdr/UpdateConfigHdr_plug_in.yaml to the environment with scope configdata
DEBUG - --- self_describing_environment.update_simple_paths()
DEBUG - --- self_describing_environment.update_extdep_paths()
DEBUG - Verify 'Mu-Basetools' returning 'False'.
DEBUG - Verify 'edk2-acpica-iasl' returning 'False'.
DEBUG - Verify 'mu_nasm' returning 'False'.
DEBUG - Verify 'mu-uncrustify-release' returning 'False'.
DEBUG - --- self_describing_environment.report_extdep_version()
DEBUG - Setting up version aggregator
DEBUG - Verify 'Mu-Basetools' returning 'False'.
DEBUG - Verify 'edk2-acpica-iasl' returning 'False'.
DEBUG - Verify 'mu_nasm' returning 'False'.
[... cut ...]
WARNING - [SDE] Failed to fetch NugetDependecy: edk2-basecrypto-driver-bin@2022.08.6: [Nuget] We failed to install this version 2022.08.6 of edk2-basecrypto-driver-bin
WARNING - [SDE] Failed to fetch NugetDependecy: Mu-Basetools@2022.08.4: [Nuget] We failed to install this version 2022.08.4 of Mu-Basetools
WARNING - [SDE] Failed to fetch NugetDependecy: mu-uncrustify-release@73.0.8: [Nuget] We failed to install this version 73.0.8 of mu-uncrustify-release
ERROR - We were unable to successfully update 5 dependencies in environment
SECTION - Summary
ERROR - Error

sunflower2333, 2023-09-18:
seems that mono and nuget is missing? try apt install nuget mono in the container.

Sunflower2333.

WARNING - [SDE] Failed to fetch NugetDependecy: ***@***.***: [Nuget] We failed to install this version 73.0.8 of mu-uncrustify-release
WARNING - [SDE] Failed to fetch NugetDependecy: ***@***.***: [Nuget] We failed to install this version 20016.1.1 of mu_nasm
WARNING - [SDE] Failed to fetch NugetDependecy: ***@***.***: [Nuget] We failed to install this version 2022.08.6 of edk2-basecrypto-driver-bin
WARNING - [SDE] Failed to fetch NugetDependecy: ***@***.***: [Nuget] We failed to install this version 2022.08.4 of Mu-Basetools
WARNING - [SDE] Failed to fetch NugetDependecy: ***@***.***: [Nuget] We failed to install this version 20210105.0.6 of edk2-acpica-iasl
WARNING - [SDE] Failed to fetch NugetDependecy: ***@***.***: [Nuget] We failed to install this version 20210105.0.6 of edk2-acpica-iasl
WARNING - [SDE] Failed to fetch NugetDependecy: ***@***.***: [Nuget] We failed to install this version 20016.1.1 of mu_nasm
WARNING - [SDE] Failed to fetch NugetDependecy: ***@***.***: [Nuget] We failed to install this version 2022.08.6 of edk2-basecrypto-driver-bin
WARNING - [SDE] Failed to fetch NugetDependecy: ***@***.***: [Nuget] We failed to install this version 2022.08.4 of Mu-Basetools
WARNING - [SDE] Failed to fetch NugetDependecy: ***@***.***: [Nuget] We failed to install this version 73.0.8 of mu-uncrustify-release

Ost268, 2023-09-18:
Sorry, problem with my ubuntu, reinstall and try it

Ost268, 2023-09-23:
root@768bd46be025:/build# stuart_ci_build -c .pytool/CISettings.py 
TOOL_CHAIN_TAG=VS2019 -a X64,IA32 -t DEBUG --verbose
SECTION - Init Self Describing Environment
ERROR - Dependency 'edk2-basecrypto-driver-bin' is not met!
ERROR - Dependency 'mu_nasm' is not met!
ERROR - Dependency 'Mu-Basetools' is not met!
ERROR - Dependency 'edk2-acpica-iasl' is not met!
ERROR - Dependency 'mu-uncrustify-release' is not met!
Traceback (most recent call last):
  File "/usr/local/bin/stuart_ci_build", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/edk2toolext/invocables/edk2_ci_build.py", line 301, in main
    Edk2CiBuild().Invoke()
  File "/usr/local/lib/python3.10/dist-packages/edk2toolext/base_abstract_invocable.py", line 215, in Invoke
    raise RuntimeError("External Dependencies in the environment are out of date. "
RuntimeError: External Dependencies in the environment are out of date. Consider running stuart_update to possibly resolve this issue.
bash: -a: command not found

sunflower2333, 2023-09-25:
what are them? have u installed pip modules ? where did you find the stuart_ci_build command?

Sunflower2333.

Ost268, 2023-09-26:
Ok thanks

Ost268, 2024-01-05:
DEBUG - Adding descriptor /build/Features/CONFIG/SetupDataPkg/Plugins/UpdateConfigHdr/UpdateConfigHdr_plug_in.yaml to the environment with scope configdata
DEBUG - --- self_describing_environment.update_simple_paths()
DEBUG - --- self_describing_environment.update_extdep_paths()
DEBUG - Verify 'mu-uncrustify-release' returning 'False'.
DEBUG - Verify 'mu_nasm' returning 'False'.
DEBUG - Verify 'Mu-Basetools' returning 'True'.
INFO - Computing path for Mu-Basetools located at /build/MU_BASECORE/BaseTools/Bin/Mu-Basetools_extdep on Host(os='Linux', arch='x86', bit='64')
DEBUG - /build/MU_BASECORE/BaseTools/Bin/Mu-Basetools_extdep/Linux-x86-64 does not exist
INFO - /build/MU_BASECORE/BaseTools/Bin/Mu-Basetools_extdep/Linux-x86 was found!
DEBUG - Verify 'edk2-acpica-iasl' returning 'False'.
DEBUG - --- self_describing_environment.report_extdep_version()
DEBUG - Verify 'mu-uncrustify-release' returning 'False'.
DEBUG - Setting up version aggregator
DEBUG - Verify 'mu_nasm' returning 'False'.
DEBUG - Verify 'Mu-Basetools' returning 'True'.
INFO - Computing path for Mu-Basetools located at /build/MU_BASECORE/BaseTools/Bin/Mu-Basetools_extdep on Host(os='Linux', arch='x86', bit='64')
DEBUG - /build/MU_BASECORE/BaseTools/Bin/Mu-Basetools_extdep/Linux-x86-64 does not exist
INFO - /build/MU_BASECORE/BaseTools/Bin/Mu-Basetools_extdep/Linux-x86 was found!
DEBUG - Verify 'edk2-acpica-iasl' returning 'False'.
DEBUG - Verify 'mu-uncrustify-release' returning 'False'.
DEBUG - Verify 'mu-uncrustify-release' returning 'False'.
ERROR - Dependency 'mu-uncrustify-release' is not met!
DEBUG - Verify 'edk2-basecrypto-driver-bin' returning 'False'.
ERROR - Dependency 'edk2-basecrypto-driver-bin' is not met!
DEBUG - Verify 'mu_nasm' returning 'False'.
DEBUG - Verify 'mu_nasm' returning 'False'.
ERROR - Dependency 'mu_nasm' is not met!
DEBUG - Verify 'Mu-Basetools' returning 'True'.
INFO - Computing path for Mu-Basetools located at /build/MU_BASECORE/BaseTools/Bin/Mu-Basetools_extdep on Host(os='Linux', arch='x86', bit='64')
DEBUG - /build/MU_BASECORE/BaseTools/Bin/Mu-Basetools_extdep/Linux-x86-64 does not exist
INFO - /build/MU_BASECORE/BaseTools/Bin/Mu-Basetools_extdep/Linux-x86 was found!
DEBUG - Verify 'Mu-Basetools' returning 'True'.
DEBUG - Verify 'edk2-acpica-iasl' returning 'False'.
DEBUG - Verify 'edk2-acpica-iasl' returning 'False'.
ERROR - Dependency 'edk2-acpica-iasl' is not met!

Ost268, 2024-01-05:
INFO - Log Started: Friday, January 05, 2024 10:13PM
SECTION - Init Self Describing Environment
DEBUG - --- self_describing_environment.__init__()
DEBUG - Skipped directories specified = ()
DEBUG - --- self_describing_environment.load_workspace()
DEBUG - Loading workspace: /build
DEBUG -   Including scopes: SurfaceDuo1, gcc_aarch64_linux, edk2-build, cibuild, configdata, global-nix, global
DEBUG - --- self_describing_environment._gather_env_files()
DEBUG - Adding descriptor /build/MU_BASECORE/edk2_core_path_env.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/basetools_calling_path_env.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/basetools_path_env.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Scripts/basetools_scripts_bin_path_env.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Source/Python/basetool_tiano_python_path_env.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/BinWrappers/PosixLike/posix_path_env.yaml to the environment with scope global-nix
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Bin/iasl_ext_dep.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Bin/basetoolsbin_ext_dep.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Bin/nasm_ext_dep.yaml to the environment with scope edk2-build
DEBUG - Adding descriptor /build/MU_BASECORE/CryptoPkg/Driver/Bin/BaseCryptoDriver_ext_dep.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/UncrustifyCheck/uncrustify_ext_dep.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/FdSizeReport/FdSizeReportGenerator_plug_in.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/FlattenPdbs/FlattenPdbs_plug_in.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/BuildToolsReport/BuildToolsReportGenerator_plug_in.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/WindowsCapsuleSupportHelper/WindowsCapsuleSupportHelper_plug_in.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/ClangPdbToolChain/ClangPdbToolChain_plug_in.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/LinuxGcc5ToolChain/LinuxGcc5ToolChain_plug_in.yaml to the environment with scope global-nix
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/OverrideValidation/OverrideValidation_plug_in.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/BaseTools/Plugin/Edk2ToolHelper/Edk2ToolHelper_plug_in.json to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/LibraryClassCheck/LibraryClassCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/GuidCheck/GuidCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/NestedPackageCheck/NestedPackageCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/ImageValidation/ImageValidation_plug_in.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/CompilerPlugin/Compiler_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/DscCompleteCheck/DscCompleteCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/RustPackageHelper/RustPackageHelper_plug_in.yaml to the environment with scope global
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/MarkdownLintCheck/MarkdownLintCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/LineEndingCheck/line_ending_check_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/SpellCheck/SpellCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/DependencyCheck/DependencyCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/UncrustifyCheck/uncrustify_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/CharEncodingCheck/CharEncodingCheck_plug_in.yaml to the environment with scope cibuild
DEBUG - Adding descriptor /build/MU_BASECORE/.pytool/Plugin/DebugMacroCheck/BuildPlugin/DebugMacroCheck_plug_in.yaml to the environment with scope global
DEBUG - Adding descriptor /build/Features/CONFIG/SetupDataPkg/Plugins/UpdateConfigHdr/UpdateConfigHdr_plug_in.yaml to the environment with scope configdata
DEBUG - --- self_describing_environment.update_simple_paths()
DEBUG - --- self_describing_environment.update_extdep_paths()
DEBUG - Verify 'mu-uncrustify-release' returning 'False'.
DEBUG - Verify 'mu_nasm' returning 'False'.
DEBUG - Verify 'Mu-Basetools' returning 'False'.
[... cut ...]
INFO - --------------Cmd Output Finished---------------
INFO - --------- Running Time (mm:ss): 00:00 ----------
INFO - ----------- Return Code: 0x00000001 ------------
INFO - ------------------------------------------------
WARNING - [SDE] Failed to fetch NugetDependecy: edk2-basecrypto-driver-bin@2023.02.2: [Nuget] We failed to install this version 2023.02.2 of edk2-basecrypto-driver-bin
SECTION - 	Updated/Verified 1 dependencies
ERROR - We were unable to successfully update 4 dependencies in environment
SECTION - Summary
ERROR - Error

Ost268, 2024-01-05:
not work this project

sunflower2333, 2024-01-06:
@Ost268 you did not install  INFO - /bin/sh: 1: mono: not found

Ost268, 2024-01-06:
ERROR - Failed to convert Edk2Relative Path to an Absolute Path on this system.
ERROR - Relative Path: SurfaceDuoFamilyPkg/Driver/UsbStartControllerSignalDxe/UsbStartControllerSignalDxe.inf

SurfaceDuoFamilyPkg/Driver/UsbStartControllerSignalDxe/UsbStartControllerSignalDxe.inf --no sush folder and file

sunflower2333, 2024-01-06:
@Ost268 remove line
SurfaceDuoFamilyPkg/Driver/UsbStartControllerSignalDxe/UsbStartControllerSignalDxe.inf
 in Platforms/SurfaceDuo1Pkg/Device/samsung-winner/DXE.dsc.inc