Project-Aloha/mu_aloha_platforms #122: Raphael not able to boot

graphiks, 2023-10-08:
I compiled edk2 myself for raphael, it shows windows 11 logo, then throws a "IRQL_NOT_LESS_OR_EQUAL" BSOD. How can i get logs for more troubleshooting?

sunflower2333, 2023-10-09:
have u installed drivers?

graphiks, 2023-10-09:
Right, i forgot. sorry.  But i cant install them , dism just throws a " error 87: Add-Driver command is unkown.

closed 2023-10-09 by graphiks

halal-beef, 2023-10-09:
@graphiks Btw we use driverupdater from the WOA-Project org to install the drivers instead of dism, I advise you to do the same

graphiks, 2023-10-10:
Can you use it while in PE? Ive tried msm on another image but it doesnt detect my phone. Can you boot into msm or anything other than windows with this image?

sunflower2333, 2023-10-10:
Yes it works ok here and it can boot linux too.

sunflower2333, 2023-10-10:
> Ok, this might be a bit of a stupid question but how do i boot in mass storage mode?

how did you apply windows image?

Sunflower2333.

graphiks, 2023-10-10:
I made a winpe partition and put the efi files on there. Windows isnt installed currently