Project-Aloha/mu_aloha_platforms #233: Vayu: the required fv image size 0x37f5f0 exceeds the set fv image size 0x300000

hiprivsid, 2024-03-02:
MU Doesn't builds for vayu giving error
```
INFO - Return Value = 2
INFO - GenFv: ERROR 3000: Invalid
INFO -   the required fv image size 0x37f5f0 exceeds the set fv image size 0x300000
INFO -
INFO -
INFO -
INFO -
INFO - build.py...
INFO -  : error 7000: Failed to generate FV
INFO -
INFO -
INFO -
INFO - build.py...
INFO -  : error 7000: Failed to execute command
INFO -
INFO -
INFO - - Failed -
INFO - Build end time: 00:39:39, Mar.03 2024
INFO - Build total time: 00:00:20
INFO -
INFO - ------------------------------------------------
INFO - --------------Cmd Output Finished---------------
INFO - --------- Running Time (mm:ss): 00:20 ----------
INFO - ----------- Return Code: 0x00000001 ------------
INFO - ------------------------------------------------
ERROR - Compiler #7000 from : Failed to generate FV
ERROR - Compiler #7000 from : Failed to execute command
```

halal-beef, 2024-03-02:
@hiprivsid you need to replace the UfsDxe with the one from LG G8X

hiprivsid, 2024-03-02:
> @hiprivsid you need to replace the UfsDxe with the one from LG G8X

But in g8x ufsdxe same

closed 2024-03-03 by hiprivsid