Project-Aloha/mu_aloha_platforms #414: Infrastructure: Duplicated ACPI Tables, impacting updates for ref platforms

gus33000, 2024-11-20:
Select QTI platforms have their ACPI files fully provided by the ACPI binary repository, due to this repo build system it is required to duplicate these files currently into the device ACPI folder with no possibility to override such requirement to instead use the file naturally available under SurfaceDuoACPI\QcomACPI\8350\builtin\DSDT_8350_MTP.aml for example.

It would be great if this could get improved for easy updates, currently we have to manually copy over each file one by one and even rename them to address this.

Further more; select devices may need to override more tables in the future, like for the debug table due to different GENI UART QUP usage for debugging (See OEM Corporation OEMZE device for a prime example of such thing)

sunflower2333, 2024-12-09:
If there is a way to override it ...

sunflower2333, 2024-12-19:
How about make a symbol link to the reference devices' acpi in reference devices' ACPI/DSDT.aml

gus33000, 2024-12-19:
I was thinking at some point given the links keep increasing, that maybe we should adopt the normal edkII/mu model to have multiple platforms here. I would be up for one package for one device. This would also resolve a few issues where a device may not have a memory map compatible enough to afford reusing the existing defacto base address in the fdf. That was a bit of the spirit in SurfaceDuo1Pkg and 2Pkg, it also contains very minimal lines in the end and you would be able to dial in the dxes you want with ease, and the acpi tables

sunflower2333, 2024-12-19:
well that make sense. But it will take time to refactor the whole repo and will reduced reusability, slow down ci, increase porting difficulty etc.
However as you say it can solve some issues currently meeting.
So i hope to find compromise solution that ensure reusability while meeting standards.

gus33000, 2024-12-19:
Build times shouldn't be an issue anymore ever since we modified the ci to build every target in parallel and not consecutively. Of course this will increase aggregated custom workflows, like the one building only ref platforms or everything 855 but those iirc are also parallelized? Or should be made to be parallel

sunflower2333, 2024-12-28:
OK i will try refactor next year. Do you have any ideas about folder structure? 

Is `Platforms/<PlatformName>/<VendorName>/<DeviceCodeName>` okay?

For example:
`Platforms/Lahaina/QTI/MTP888Pkg`

sunflower2333, 2025-02-24:
Test bot.

gus33000, 2025-02-24:
I do believe having the soc codename is redundant as as a platform you are going to take dependency on something under silicon anyway.

I propose this instead:

```Platforms/Qualcomm/MTP8350Pkg/MTP8350Pkg.fdf``` etc

BTW: Some personal nits:

- Prefer Qualcomm over QTI, as it is more explicit
- Prefer 8150 over 855 (rename was official by that point)

sunflower2333, 2025-02-25:
> ```Platforms/Qualcomm/MTP8350Pkg/MTP8350Pkg.fdf``` etc

This form perhaps good for qti devices. But for other devices it will be mess i think. 
That's why i say add a redundant folder under Platform.(it's just a folder for sorting and no other files there)

sunflower2333, 2025-02-25:
> - Prefer Qualcomm over QTI, as it is more explicit
> - Prefer 8150 over 855 (rename was official by that point)
> 

okay.

gus33000, 2025-02-25:
Im drafting up a layout for a non made by Qualcomm Technologies Inc. series of devices to explain my thought process here:

First, here is what I am suggesting:

```
+-Platforms
+-+-OnePlus
| +---OnePlus5Pkg
| +---OnePlus6Pkg
| +---OnePlus6TPkg
| +---OnePlus7Pkg
| +---OnePlus7ProPkg
| +---OnePlus7TPkg
| +---OnePlus7TProPkg
...
```

Second, here is your proposal:

```
+-Platforms
+-+-OnePlus
| +-+-MSM8998
| | +---OnePlus5Pkg
| |
| +-+-SDM850
| | +-OnePlus6Pkg
| | +-OnePlus6TPkg
| |
| +-+-SM8150
| | +-OnePlus7Pkg
| | +-OnePlus7ProPkg
| | +-OnePlus7TPkg
| | +-OnePlus7TProPkg
...
```

I honestly do not see how it would start being a mess, looks quite reasonable to me on my first example. Everyone also gets what their device is, I would even go as far as potentially say that probably most people wouldnt know what soc they use more than their device...

gus33000, 2025-02-25:
I mixed up your proposal so redoing it again with the right layout (it doesnt change what I think, i potentially think both ways dont really provide a major help, hw manufacturers dont make thousands of diff phone models, and im seeing a ton of empty folders otherwise)

+-Platforms
+-+-MSM8998
| +-+-OnePlus
| | +---OnePlus5Pkg
| |
+-+-SDM850
| +-+-OnePlus
| | +---OnePlus6Pkg
| | +---OnePlus6TPkg
| |
+-+-SM8150
| +-+-OnePlus
| | +---OnePlus7Pkg
| | +---OnePlus7ProPkg
| | +---OnePlus7TPkg
| | +---OnePlus7TProPkg
...

sunflower2333, 2025-02-25:
For some vendors(like poco/redmi/realme), it uses different SoC in same series like will make the devices under <Vendor> quite confusing.

As a exmaple:
PocoX3 Pkg-> SM7150
PocoX3Pro Pkg -> SM8150
or:
Realme X3 -> SM8150
Realme X3 Pro -> SM8250
Realme X3 Superzoom -> SM8150
or: 
Redmi K80 -> SM8650
Redmi K80 Pro -> SM8750

Not everyone can&will remember all this relationships. When there is an platform issue being found and needs fix (like replacing a dxe inf for in devcies' fdf, or memory map region optimize for all devices under same platforms),
i can't image how terrible it will be:   
   You have to find out all device pkgs under a same platform, search their name on google one by one or grep platform strings in <vendor>/ or read docs to them out and edit their fdf(s) /dsc{s}/map(s) to fix a platform-general issue.

gus33000, 2025-02-25:
I mean wouldn't in this case the problem also occur with your layout? People will confuse it too if they don't know how their device is named

sunflower2333, 2025-02-25:
Which layout you mean, the current one or assumed one

gus33000, 2025-02-25:
Any including my suggestion. There is so much you can do when the user doesn't know his phone is called Poco X3 Pro and not Poco X3 here. Even with your above structure, they can totally pick the wrong one as the issue is an issue where the user has no idea about their own device. I don't see how adding soc in the path helps in any way. I find it more painful too as there's only going to be few folders here anyway.

sunflower2333, 2025-02-25:
It's not a user side issue.

In a view of a developer. without a folder of <chip-platform> will made the repo hard to be maintain when a a single platform issue is found for all devices under the chip plaform. 

And this problem often happens.  Do you still remember last time the logo disappeared on sm8350 devices? It's caused by GraphicConsoleDxe is not the stock one. I replaced GraphicConsoleDxe .inf in all sm8350 devices. There is an LahainaPkg currently so i can easily edit DXE.inc of all sm8350 devices in one command 
```
vim LahainaPkg/devices/*/DXE.inc
```
In my new layout it just break big pkg(lahainaPkg) into small pkgs(Oneplus9Pkg), when similar issue happends, developers can easily do fix in single command to edit all files or check these files in IDE under the folder.
```
vim Lahaina/*(Vendor)/*(Device)/*.fdf
```
Without a folder of <chip-platform> developers have to find the sm8350 devices one by one in IDE/cmd interface. That makes thing mess.

sunflower2333, 2025-02-25:
Anyway, I'll take your suggest to not make you feel painful.

sunflower2333, 2025-02-25:
Currently separating binaries is WIP. 
I plan to do all "unpack" pkg & separate binaries works by myself. It will be done in a month.

sunflower2333, 2025-02-25:
Will create a PR after finishing.

sunflower2333, 2025-02-25:
It is both a device specific thing and a platform specific thing. You can not move it in to chip package. I'm just giving an example.
I also mentioned memory map above, it is a typically device specifical and platform specifical thing.

sunflower2333, 2025-02-25:
memory map can not be share in chip package, unless you real like a map full of macro patches like edk2msm

gus33000, 2025-02-25:
To be clear, I agree with you above and my issue written above was also never about changing the folder structure. That wasn't the issue at play and I don't think changing the structure does anything to fix the issue

gus33000, 2025-02-25:
The only thing having specific package does is being able to reuse them and override what you want properly. Right now there's a single package for every device in the world. That's chaotic and prevents properly changing settings per device. But it doesn't solve above.

sunflower2333, 2025-02-25:
>  fix the issue
At least it won't let thing go worse, if the issue you mean here is `the reusble contents' complicated/redundant modification for each device`.

gus33000, 2025-02-25:
We need a way to further enable overrides, but the issue here is about ACPI. I was simply making a suggestion about adopting packages for each devices to fix this (which can be done without changing the structure). Having a package per device enables you to define exactly the paths you want afterall. Its not the sole solution however.

sunflower2333, 2025-02-25:
And it can fix this issue perfectly

sunflower2333, 2025-02-25:
just give a sort:
```
First priority is fixing acpi duplicated and not all table are device specifical. 
 -> Then you said can use small pkgs
    -> I accept this suggestion because i also want to unpack current Big packages.
    -> I give a new layout, request for comments
        -> you say <platform name> folder is not neccessary
            -> we discussion about its necessity 
               -> conclusion: current stat many thing not reusable.  Simply add a <platform name> folder can't fix.

 -> back to first priority
```

sunflower2333, 2025-02-25:
> The simplest form of a fix to my eyes (i dont deny theres others) would involve pretty much taking everything you have today under lets say `\Platforms\NazgulPkg`, and copying it under `\Platforms\NazgulPkg\Device\*\`, deleting all content but "devices" under `\Platforms\NazgulPkg`, and then cleaning things up.
> 
> But restructuring I think is a completely different topic open for debate and I do not think, that any of our proposal for a potential restructure, is good at the moment.

For simply fixing ACPI tables use are not very flexible. Just move acpi.inc into each device's folder will be ok

sunflower2333, 2025-02-25:
Add a extra ACPI.inc under <device-codename>, accompany DXE,inc, DXE.dsc.inc, APRIORI.inc, etc under that folder.

gus33000, 2025-02-25:
This indeed is the simplest to fix acpi but let's say I also want a different way to package/build the boot partition and it falls apart again (and I wish we could too btw it is inconvenient to have v0 especially when you want to edit the EFI quickly

sunflower2333, 2025-02-25:
@gus33000 The pack part is written in python and is easy to eXtend more functions.

A possible way: 
1. write a flag "Define BOOT_HDR_VER = xxx" in Defines.dsc.inc under <vendor-codename>/

2. let the script(PostBuild.py) parse Defines.dsc.inc directly with strings processing, or read from edk2 build sysyem  to get value of the macro.

3. Pack the boot image as what defined in that macro.

4. Add the macro usage in our docs.

5. Done

gus33000, 2025-02-25:
Now what if I want to have a different loading address for the uefi fd? I mean even in the boot image alone there's so many parameters too just specifying the version wouldn't be enough. And in my view adding more and more variables feels super hacky versus just giving the user its own pkg which would also look cleaner on top of offering the flexibility that was needed

remtrik, 2025-02-25:
@gus33000 Talking about different uefi fd address, that's exactly what all lge 8150 devices require

sunflower2333, 2025-02-25:
@remtrik why

sunflower2333, 2025-02-26:
@gus33000 At least you need to write the boot version(or other parameters) at somewhere.
And for flexibility, give it a default value without writing anything.  In my view a config file is another type of 'macro'. 

Move these parameters into each device's fdf/dsc can not help change the fact of defining macros.🙃

remtrik, 2025-02-26:
@sunflower2333 LGE made tgcm so big it's actually 3-4 times bigger when compared with other devices, so it will overlap currently used uefi fd and other regions near it

sunflower2333, 2025-02-26:
@remtrik You mean  it has a different uefi fd region from other devcies?

remtrik, 2025-02-26:
@sunflower2333 Yes, almost all hardcoded in fdf/dsc regions address are different

sunflower2333, 2025-02-26:
> [@sunflower2333](https://github.com/sunflower2333) Yes, almost all hardcoded in fdf/dsc regions address are different
No, it isn't
[image]

gus33000, 2025-02-26:
> > [@sunflower2333](https://github.com/sunflower2333) Yes, almost all hardcoded in fdf/dsc regions address are different
> No, it isn't
> [image]

Uefiplat.cfg in xbl for la platforms is almost never correct and doesn't have to be. It's only purpose is so Linux loader loads, whenever regions are correct or not never matters and nobody even QC edits those correctly because it's all overwritten by the DTS anyway. Linux loader doesn't take into consideration any EFI map at all

gus33000, 2025-02-26:
> @gus33000 At least you need to write the boot version(or other parameters) at somewhere.
> And for flexibility, give it a default value without writing anything.  In my view a config file is another [...]

But now that means I am forced to use whatever script built in here exists and the moment I want a different format then I have to submit a pr just to add said format nobody uses or I could have had my own simple file from the start to do it on my own on my own pkg.... (Like so: https://github.com/WOA-Project/mu_andromeda_platforms/blob/main/build_boot_epsilon.sh

gus33000, 2025-02-26:
btw a real world example of a problematic situation requiring a completely different approach for building boot images: https://github.com/Project-Aloha/mu_aloha_platforms/tree/kona/dev/devices/qrd/Platforms/KonaPkg/Device/qcom-qrd8250/ImageResources

MTP is also affected I just never did the same as this for it. Will revisit tomorrow and make a PR for the MTP as well. I do plan on pushing more 8250 stuff next week anyhow.

sunflower2333, 2025-02-27:
@gus33000 i feel tired about discussing how to solve this issue and i need time to think. 
Will continue replying next week.