Project-Aloha/mu_aloha_platforms #447: Dubious Device Support, Xiaomi-Sweet

gus33000, 2025-01-19:
Following some internal conversations on telegram; it appears xiaomi-sweet is a device that is using the SM7150 system on chip, yet was added under this repository claiming it used SM7125/SM6250 (or one of its derivatives).

This is a problem; first because of it being under the wrong platform folder
And second because it is a platform we do not support at all at the moment.

Others also claim said uefi port doesnt even work at the moment.

Following the decision being actively taken at the moment as part of https://github.com/Project-Aloha/mu_aloha_platforms/pull/445 an action needs to be taken as a direct following of above decision that is one of the following:

- Deleting said device support entirely
- Moving said device support to the corresponding SM7150 platform support
- Moving said device support to a secondary repository

gus33000, 2025-01-19:
Final decision is to support SM7150, need to create a PR later to move that device

sunflower2333, 2025-01-22:
I removed it currently. It can still be recovered from git history. Will recover until next time find someone test/dev it.