Project-Aloha/mu_aloha_platforms #448: Regression: Every Qualcomm Reference Platforms report "Unknown" as device model

gus33000, 2025-01-19:
Self explanatory. Should be reporting MTP/QRD/ATP/IDP/whatever instead, but says unknown. Need to investigate.

sunflower2333, 2025-01-20:
I’m going to optimze the logic. If a Package does not contains a reference device, build a single random device in it. 
The purpose of this workflow is for checking if a push/merge break building for each pkg.

sunflower2333, 2025-01-20:
oh, i thought you mean CI bug

gus33000, 2025-01-20:
Very rushed and quick analysis so don't take my words for it as it may not be correct but from a quick glance, we check if the predefined pcd is of length higher than 0 and if it is we treat it as an override and don't call into QC routines. This is good but given we preset everything to a string "Unknown" it also makes the code never run. We should instead preset these vals to an empty string I think.

sunflower2333, 2025-01-20:
[image]
You mean here? Yep the predefined value in dec will cause the issue

gus33000, 2025-01-20:
I am at work at the moment I'll check later on my devices if it's fixed if nobody tried it yet on a QC device that is. Thanks!

gus33000, 2025-01-20:
[image]

looks fixed here

closed 2025-01-20 by gus33000