Project-Aloha/mu_aloha_platforms #451: SMBIOS: Baseboard product is not correctly updated

gus33000, 2025-01-26:
- Device Under Test: HDK8150
- UEFI version: latest commit as of this issue
- Tools used for testing: UFPTool

- Boot Device Using the latest MTP uefi file
- Device should automatically go into UFP mode due to no EFIESP/ESP partition on the device (this just has android on it)
- use UFPTool to collect flash logs and open them
- Notice below output:

 ```
[00:08:08] 0045 INFO: HandleInfoQueryReq: Qualcomm.SDM855.HDK.2.2
[00:08:08] 0046 INFO: HandleInfoQueryReq: Async support: 1
[00:08:08] 0047 INFO: SecGetUefiSecureBootStatus: Read UEFI secure boot Status
[00:08:08] 0048 INFO: SecGetUefiSecureBootStatus: SECURITY_UEFI_SECURE_BOOT: Status 0x00
[00:08:08] 0049 INFO: HandleInfoQueryReq: PSB: 0xff
[00:08:08] 004a INFO: HandleInfoQueryReq: SFFU: 0xff
[00:08:08] 004b INFO: HandleInfoQueryReq: USB: 0x00
[00:08:08] 004c INFO: HandleInfoQueryReq: Supported SFFU protocols: 0x000f
[00:08:08] 004d INFO: HandleInfoQueryReq: CRC header info: Version: 0x00000001, Alignment: 0x00000004
[00:08:08] 004e INFO: HandleInfoQueryReq: Device Target Info...
[00:08:08] 004f INFO: HandleInfoQueryReq: Manufacturer: Qualcomm
[00:08:08] 0050 INFO: HandleInfoQueryReq: Family: SDM855
[00:08:08] 0051 INFO: HandleInfoQueryReq: Product Name: HDK
[00:08:08] 0052 INFO: HandleInfoQueryReq: Product Version: 2.2
[00:08:08] 0053 INFO: HandleInfoQueryReq: SKU Number: 6
[00:08:08] 0054 INFO: HandleInfoQueryReq: Baseboard Manufacturer: Qualcomm
[00:08:08] 0055 INFO: HandleInfoQueryReq: Baseboard Product: MTP
```

The SMBIOS Code is meant to dynamically detect the underlying platform. This works as expected as shown in the PLAT ID and product Name being returned (both set as HDK). However; the baseboard product is not correctly updated and is hardcoded to MTP. This should not be the case on reference implementations

gus33000, 2025-01-26:
This issue remains low impact as the real field checked in order to forge the PLAT ID is product name which does get updated. And the baseboard product field is not exactly used by anything important and remains most cosmetic/informational than deal breaking.

(A different plat id would cause blockage when flashing FFU files unless using dangerous overrides if it impacted its generation, however this is not the case and things flash fine without the use of any override:

```batch
UFPTool.exe FlashDevice -Path "D:\GitHub\WOA-Project\Main\windows_qcom_platforms\tools\Firmware\FFUs\QCOM_HDK_8150_MaximizedForWindows.ffu"
```

[image]
)

gus33000, 2025-01-27:
The issue is located here: https://github.com/Project-Aloha/mu_aloha_platforms/blob/main/Platforms/SurfaceDuoFamilyPkg/Driver/SmBiosTableDxe/SmBiosTableDxe.c#L785

We make use of the pcd without any override as planned. We need to keep such pcd for other devices but need to use the platform info on ref platforms, (potentially done in the same way, if pcd blank then use the standard api value which would give us "HDK")

sunflower2333, 2025-02-03:
The value is same with PcdSystemRetairModel on Reference Platforms ?

gus33000, 2025-02-03:
This should not be hardcoded and be dynamically retrieved. Thats the whole point of the already existing code here. The PCD is an override for third party OEM devices instead.

commit ca0ec7e2 (2025-02-04): SMBIOS: Dynamic Reg Platform Logic Updates, fixes #451
closed 2025-02-04 by gus33000