Project-Aloha/mu_aloha_platforms #480: 8450 is not 8475, Incompatibility issue and confusion with Palima versus Waipio

gus33000, 2025-02-27:
Currently we have a waipio device (hdk8450) defined under Palima. Early testing on MTP8450 (back when I could power it on >.>) showed that Palima code really is not compatible with Waipio. (for context, QC own ref devices always share the same code so...)

Also it is uncertain if HDK8450 really is a waipio device (8450) or a Palima device (8475)

Maybe we have to look into above? I lack the HDK to know what it really is myself.

sunflower2333, 2025-02-28:
It works on hdk8450.
I think i should remove all sm8475 stuff and only reserve sm8450 test devices(dev board or mtp etc.) for only testing use.

sunflower2333, 2025-03-01:
Patched Binaries for HDK8450 are dumped from stock fw, not 8475's binaries.
[image]

gus33000, 2025-03-01:
Maybe firmware version difference? I cant really look into it anyway because I dont have a debug board that works for MTP 8450 anyway :/

gus33000, 2025-03-01:
Yes but how do I power on the device?

gus33000, 2025-03-01:
The device still needs electricity to work lol

sunflower2333, 2025-03-02:
i see lol.

closed 2025-04-10 by sunflower2333

gus33000, 2025-04-27:
> The device still needs electricity to work lol

fixed

gus33000, 2025-04-27:
@sunflower2333 just to be clear you have a HDK8450 not HDK8475 right?

sunflower2333, 2025-04-27:
@gus33000 Yes.