Project-Aloha/mu_aloha_platforms #483: SecureBoot: SiPolicy update issues for special circumstances

qaz6750, 2025-03-01:
## Brief description
* In the current SecureBootProvisioningDxe, when the SiPolicy already exists, even if the SiPolicy is modified, updated, the SiPolicy in the ESP partition of the user device will still remain the same. Of course, under normal circumstances, the SiPolicy does not require any modification or update. Here, we are only discussing the special cases where the SiPolicy is changed or updated.

gus33000, 2025-03-01:
There is a bug in EDKII/Mu FAT FileSystem code that will cause FAT32 FileSystem corruptions to the FATs in the partition and thus corrupt it. Thats why at some point I added some piece of code to not rewrite it otherwise a lot of end users complained of their boot being stuck after a few reboots (due to said corruption)

gus33000, 2025-03-01:
Currently the only time we touch the file is if you boot with secureboot off (as you can see in your snippet) to delete the file. I am not entirely sure how to build something here to update it in case (apart maybe hashing it?) it is my understanding its only the writes that cause corruption, but i guess hashing the file on disk and comparing it with the hash of the one in our FV would work

qaz6750, 2025-05-02:
Perhaps the method of calculating the hash of SiPolicy can be used to determine whether the UEFI needs to update SiPolicy. However, since the relevant methods of SHA1 and MD5 have been abandoned by MU, and I don't want this function to make the UEFI loading process take longer, it may be necessary to use other hash calculation methods.

closed 2025-06-14 by qaz6750