Project-Aloha/mu_aloha_platforms #521: UFS3 setting is not dynamic enough

sunflower2333, 2025-04-12:
It’s hardcoded and default to ufs3 in AndromedaPkg DEC and i don't know why. 
Many old devices has ufs2 and default set to ufs3 will cause issue.
Please use a more dynamic way to get ufs version and set instead of hardcoding in dsc/dec.

halal-beef, 2025-04-12:
Hardcoded to UFS3 because:

UFS3 Device booting with UFS2 Spec -> Boots -> Overvolting UFS -> Future issues

UFS2 Device booting with UFS3 Spec -> Doesn't boot -> Switch PCD

its hardcoded purely for safety

qaz6750, 2025-04-12:
It seems that on the currently mainstream-supported HANA platform, there is no way （dynamic way） to figure out whether the UFS chip is UFS 3 or UFS 2. Therefore, is it necessary to use the method of hard coding?

halal-beef, 2025-04-12:
iirc there is a reg you can read to get UFS version and it'll return 0x300 for UFS 3 and whatnot but to read it im pretty sure youd need the ufs to be powered up

sunflower2333, 2025-04-12:
like burn or something else

remtrik, 2025-04-12:
> have you encontered any over voltage issue?

My 10 vayus died because of that. I thought it was a motherboard issue, but after noticing that all devices use ufs2 settings and disassembling vayus I found 8 out of 10 ufs chips being completely fried(other 2 are barely working)

remtrik, 2025-04-12:
Regarding as to why it's hardcoded to UFS3, it was stated in the initial [PR](https://github.com/Project-Aloha/mu_aloha_platforms/pull/454#issue-2821248109)

sunflower2333, 2025-04-12:
> > have you encontered any over voltage issue?
> 
> My 10 vayus died because of that. I thought it was a motherboard issue, but after noticing that all devices use ufs2 settings and disassembling vayus I found 8 out of 10 ufs chips being completely fried(other 2 are barely working)

~~sound normal on vayu~~ 
I just checked ACPI and only 0.4v voltage difference. 
UFS3 VCC is 2.5V, UFS 2 VCC is 2.9V

As what UFS3.1 spec say:
[image]

VCC has 2 different ranges. It's up to the vendor anyway.

remtrik, 2025-07-20:
https://github.com/Project-Aloha/mu_aloha_platforms/pull/454#issuecomment-2625013269

Returning to this theme, I wonder is this true for both WP and LA or just the WP. Because 2023 Moorea BSP doesn't have that