Project-Aloha/mu_aloha_platforms #535: Review, fix, or add/enable support for new targets

gus33000, 2025-04-27:
I will personally test/handle this this very week:

- MTP8450
- MTP8550
- HDK8350
- HDK8550
- QRD7350
- QRD8x50

also this target in an out of branch (for sanity verification, this wont be intended to be in main at all but its a good comparison tool)

- SCP_MAKENA (cold boot)

Got all of the hw required to start on this as of today.

gus33000, 2025-04-27:
This will also be the occasion to publish binaries and partitions (as we now want for rather obvious reasons) as well!

sunflower2333, 2025-04-28:
> I will personally test/handle this this very week:
> 
> * MTP8450
> * MTP8550
> * HDK8350
> * HDK8450
> * QRD7350
> * QRD8x50
> 
> also this target in an out of branch (for sanity verification, this wont be intended to be in main at all but its a good comparison tool)
> 
> * SCP_MAKENA (cold boot)
> 
> Got all of the hw required to start on this as of today.

That's crazy .

gus33000, 2025-05-06:
ok so slight change in plan i will but later (not previous week heh), just too busy with work atm