Project-Aloha/mu_aloha_platforms #598: [Plan] Upgrade mu_basecore Version to release/202502

qaz6750, 2025-08-30:
## Brief description
The upstream of mu_basecore has officially entered the development phase of the 202502 branch quite some time ago and has already released the 202502 version. Meanwhile, the development of the 202405 branch is gradually falling behind. Therefore, I am now proposing the idea of migrating to the release/202502 branch and hope to get some suggestions and conduct discussions on this matter.

gus33000, 2025-09-03:
Successful migration on andromeda platforms for reference: https://github.com/WOA-Project/mu_andromeda_platforms/compare/858cb27c84dd8d9d113453559344007a33d91286...main

the PrePI rework is definitely dirty though and doesnt follow the current replacement offered in ArmPkg, also the timer constructor shouldnt be dummy i think compared to before? but this works for now

qaz6750, 2025-09-07:
> Successful migration on andromeda platforms for reference: [WOA-Project/mu_andromeda_platforms@858cb27...main](https://github.com/WOA-Project/mu_andromeda_platforms/compare/858cb27c84dd8d9d113453559344007a33d91286...main)
> 
> the PrePI rework is definitely dirty though and doesnt follow the current replacement offered in ArmPkg, also the timer constructor shouldnt be dummy i think compared to before? but this works for now

Thank you very much. I will migrate based on this reference next weekend and bring up the PR to ensure our safe transition to the new branch for all platforms.